mwood77/winderoo #11: mDNS stops working after ~2 minutes

Anumann, 2023-09-07:
I'm basically seeing this issue - any idea if there's an easy fix?

https://github.com/espressif/arduino-esp32/issues/4406

mwood77, 2023-09-15:
I had a quick look at this, but I'm unable to debug it as I'm away from my dev station for a week or so.

The base library `ESPmDNS` used in Winderoo has a comment about calling an `update()` method in the primary loop (it would be called as `MDNS.update()` at line 600 in Winderoo's `main.cpp`). However, I don't see this method existing in the library.

Are you using a Windows PC to when you try to resolve `winderoo.local`?

Anumann, 2023-09-15:
Yeah, I'm using a Windows PC. I can also test with an Android phone, if that would be helpful to test or rule anything out.

mwood77, 2023-09-16:
give it a shot with your android phone too - I remember there may have been an issue with them. I can look at it again when I'm back in the country.

Anumann, 2023-09-21:
@mwood77 sorry for the delayed response. I did some more testing - I cannot resolve the webpage winderoo.local on either my Android (using the local wifi) or my Windows PC.

One more test which also suggests to me DNS isn't working as expected - I started a ping in Windows right when the system powered on (in windows cmd window, I ran "ping winderoo.local -t"). Since the DNS resolution was working initially, the pings continued successfully even when the webpage started to fail to load.

mwood77, 2023-10-10:
Interesting - I'll look at it this week and get back to you

closed 2023-11-02 by mwood77