mwood77/winderoo #42: Wifi setup not working

Lu-ni, 2024-03-19:
Hey!
Firstly, thanks a lot for the project, I'm currently building one.

I'm encountering a issues when I follow the step to setup the winderoo.

-  DONE : Connection to winderoo wifi
-  DONE : selecting my wifi and putting the password (tried on 3 different wifi)

After those step I'm stuck on the page "Saving Credentials \n Trying to connect ESP to network ...."

Do you have any clue on how I can debug that ?

Lucas

mwood77, 2024-03-20:
Hey Lucas!

This can take some time - if you leave it for 60 seconds or so after you see "Saving Credentials...." on your phone, does the second (blinking) LED on the Winderoo board turn off? If so, that means it has successfully connected.

Also, how far away from your access point is the Winderoo? These boards don't have amazing reach, but it's usually strong enough to span some distance (I can reach across 2 floors in my wooden-structured home).

Lu-ni, 2024-03-21:
Thank you for your fast answer, I'll try again tomorrow !

closed 2024-04-01 by mwood77