mwood77/winderoo #62: Add Home Assistant Integration

TomYoungs, 2024-08-17:
Hey!

This might already be possible but would be nice to be able to control it via a home assistant dashboard, see when it's running etc, would fit into peoples automations as well which would be cool

Thanks, 
Tom

mwood77, 2024-08-19:
Interesting idea - what kind of features would you expect from that?

You could actually interact with winderoo via home assistant automations right now, if you use the [`RESTful Command` integration](https://www.home-assistant.io/integrations/rest_command/).

For example, I think this is all you'd need to get yourself up and running:

```yaml
rest_command:
  winderoo_change_timer:
    url: "http://winderoo.local/api/timer"
    method: POST
    payload: "timerEnabled={{ timerEnabled }}"
    content_type: "application/x-www-form-urlencoded"

  winderoo_change_state:
    url: "http://winderoo.local/api/update"
    method: POST
    content_type: "application/json"
    payload: >
      {
        "tpd": "{{ tpd }}",
        "hour": "{{ hour }}",
        "minutes": "{{ minutes }}",
        "timerEnabled": "{{ timerEnabled }}",
        "action": "{{ action }}",
        "rotationDirection": "{{ rotationDirection }}",
        "screenSleep": {{ screenSleep | lower }}
      }

  winderoo_toggle_power:
    url: "http://winderoo.local/api/power"
    method: POST
    content_type: "application/json"
    payload: '{"winderEnabled": "{{ winderEnabled }}"}'

  winderoo_reset:
    url: "http://winderoo.local/api/reset"
    method: GET

  winderoo_get_status:
    url: "http://winderoo.local/api/status"
    method: GET
    content_type: "application/json"
```

Then you'd call it in an automation like:

```yaml
# my automation

automation:
- alias: "Arrive at Work"
  trigger:
    platform: zone
    entity_id: device_tracker.my_device
    zone: zone.work
    event: enter
    action:
      - service: rest_command.winderoo_change_state
        data:
          tpd: "330"
          hour: "14"
          minutes: "30"
          timerEnabled: "1"
          action: "START"
          rotationDirection: "BOTH"
          screenSleep: false
    mode: single # you may or may not need this "mode: single" part
```

mwood77, 2024-10-07:
@TomYoungs proper home assistant integration is coming over MQTT. Most of the work required has been completed. Will likely release it in the next 48 hours.

see #66

mwood77, 2024-10-08:
✅  Home Assistant is available as of release 3.0.0 (main branch)