mwood77/winderoo #64: Winder only runs CCW regardless of setting

markbyrne, 2024-09-10:
I have set CW, 800 with a start time of 10:00, but noticed it was running CCW. When I manually start and stop it also only runs CCW. It was working properly yesterday...

mwood77, 2024-09-12:
That's strange. Have you refreshed the page on `Winderoo.local` to make sure the GUI is up to date?

Also, you can query winderoo's state directly with:

[winderoo.local/api/status](winderoo.local/api/status)

I'm away from my winders right now, but the response should have a field called "direction" and the possible values that it could return are: `CW | BOTH | CCW`

burgosgts, 2025-01-31:
Good afternoon, first of all, thank you for the great work. I have the same problem, the motor only turns CW. Regardless of the mode I put it in, it recognizes the three modes but only turns clockwise.

markbyrne, 2025-02-17:
This issue had resolved, but I just got back from a tip and the issue re-emerged. 

{"status":"Winding","rotationsPerDay":"800","direction":"CW","hour":"12","minutes":"00","durationInSecondsToCompleteOneRevolution":8,"startTimeEpoch":1738449903,"currentTimeEpoch":1738449933,"estimatedRoutineFinishEpoch":1738462603,"winderEnabled":"1","timerEnabled":"1","db":-69,"screenSleep":false,"screenEquipped":false}

It appears CW should be set, but it is spinning CCW.

mwood77, 2025-02-18:
Hi all do you know what version you've installed?

The front end and the back end are both versioned;

## Backend
- If you have an OLED screen attached, it'll flash the version on boot. If not, you can open up the source code and see what it is at this line:
https://github.com/mwood77/winderoo/blob/main/src/platformio/osww-server/src/main.cpp#L97

## Frontend
- Tap "Reset" and at the bottom of the pop-up, the version is listed there.

markbyrne, 2025-02-18:
Looks like 1.0.0

mwood77, 2025-02-19:
Could you please pull the code again (download the code in the `main` branch) and re-flash your microcontroller?

markbyrne, 2025-03-06:
I just re-flashed, reconnected it to wifi, set the settings, saved, hit run, and it ran in CW for ~5 seconds, then reversed and only runs in CCW again regardless of my settings.

markbyrne, 2025-03-07:
{"status":"Winding","rotationsPerDay":"800","direction":"CW","hour":"10","minutes":"00","durationInSecondsToCompleteOneRevolution":8,"startTimeEpoch":1740765449,"currentTimeEpoch":1740765462,"estimatedRoutineFinishEpoch":1740778149,"winderEnabled":"1","timerEnabled":"1","db":-66,"screenSleep":false,"screenEquipped":false}

 
[image]

It is winding CCW.

mwood77, 2025-03-19:
@markbyrne I've finally had some time to sit down with this.

I've completely wiped my test setup, re-built the frontend, and flashed the microcontroller with the code in `main`.

Versions:
- `main.cpp` is `String winderooVersion = "3.0.0";`
- Frontend is `1.1.0`

<hr />

### Direction is set to `BOTH`

API `/status` response:
```
{
    "status": "Winding",
    "rotationsPerDay": "220",
    "direction": "BOTH",
    "hour": "00",
    "minutes": "00",
    "startTimeEpoch": 1162,
    "currentTimeEpoch": 1812,
    "estimatedRoutineFinishEpoch": 3057,
    "winderEnabled": "1",
    "timerEnabled": "0",
    "db": -43,
    "screenSleep": false,
    "screenEquipped": true,
    "customWindDuration": "180",
    "customWindPauseDuration": "15",
    "customDurationInSecondsToCompleteOneRevolution": 8
}
```

✅  In this state, winding direction works correctly; alternates.

<hr /> 

### Direction is set is `CW`

API `/status` response:
```
{
    "status": "Winding",
    "rotationsPerDay": "220",
    "direction": "CW",
    "hour": "00",
    "minutes": "00",
    "startTimeEpoch": 1837,
    "currentTimeEpoch": 1981,
    "estimatedRoutineFinishEpoch": 3732,
    "winderEnabled": "1",
    "timerEnabled": "0",
    "db": -44,
    "screenSleep": false,
    "screenEquipped": true,
    "customWindDuration": "110",
    "customWindPauseDuration": "15",
    "customDurationInSecondsToCompleteOneRevolution": 8
}
```

✅  In this state, winding direction consistently starts & stops, and **only winds clockwise**

<hr />

### Direction is set is `CCW`

API `/status` response:
```
{
    "status": "Winding",
    "rotationsPerDay": "220",
    "direction": "CCW",
    "hour": "00",
    "minutes": "00",
    "startTimeEpoch": 2156,
    "currentTimeEpoch": 2165,
    "estimatedRoutineFinishEpoch": 4156,
    "winderEnabled": "1",
    "timerEnabled": "0",
    "db": -44,
    "screenSleep": false,
    "screenEquipped": true,
    "customWindDuration": "110",
    "customWindPauseDuration": "15",
    "customDurationInSecondsToCompleteOneRevolution": 8
}
```

✅ In this state, winding direction consistently starts & stops, and **only winds counter clockwise**

<hr />

Are you _certain_ everything is wired correctly on your unit?

mwood77, 2025-04-29:
Hey @yannicschuller  - thanks for letting me know. Can you dump your `platform.ini` file here please?

mwood77, 2025-04-29:
- Are you using the default motor controller (L298N) or a different motor controller?
- Is your BE version `4.0.0`? See this line: https://github.com/mwood77/winderoo/blob/216bb8400f6aa64514aeed6e34185d53559fc952/src/platformio/osww-server/src/main.cpp#L109
- Is your FE version `2.0.0`? See this line: https://github.com/mwood77/winderoo/blob/216bb8400f6aa64514aeed6e34185d53559fc952/src/angular/osww-frontend/package.json#L3

--

- Are you changing the motor direction through Home Assistant, or through the webapp? 
    - The Home Assistant state will ("should") automatically update when Winderoo acknowledges the request. The webapp's state is only "fetched" on a refresh. For example, if you change the winding direction in HA, then you need to refresh the webapp (reload the page in the browser).