mwood77/winderoo #65: 404 - Resource Not Found

whatthefilament, 2024-09-27:
I just went through setup, everything seemed to go fine. When i finished connecting to wifi it rebooted and appeared normal. Tried Navigating to Winderoo.local and got an error message 404 - Resource Not found. 

I tried erasing the flash memory
[image]
[image]
, rebuilding the firmware and reinstalling but got the same thing.

mwood77, 2024-09-28:
Hi!

This is a symptom of the file system image missing. There are two upload steps when you flash the microcontroller. If you look the image on step 5 here:

https://github.com/mwood77/winderoo/blob/main/docs/install-software.md#flashing-your-microcontroller

And then perform step 7, you should be all fixed!

whatthefilament, 2024-09-28:
hahaha, my god, how did I miss that step 7 was a different upload. Thank you. That fixed it immediately and I am up and running!

closed 2024-09-28 by whatthefilament