mwood77/winderoo #68: Stepper motor support

mwood77, 2024-10-14:
### Discussed in https://github.com/mwood77/winderoo/discussions/61

<div type='discussions-op-text'>

<sup>Originally posted by **anjanbabu** August 12, 2024</sup>
This is an amazing project! love that it has a phone UI and OLED support! 
I see a lot of DIY watch winders use small stepper motors like the [28BYJ-48 unipolar stepper motor with the ULN2003 motor driver](https://randomnerdtutorials.com/esp32-stepper-motor-28byj-48-uln2003/), it would be nice if you could enable the Winderoo code to work with these motors as well since they're so cheap and accessible. These fancy 3D printable gyro winders also use the same motors. 
I did attempt to modify the code as [suggested](https://esp32.com/viewtopic.php?p=135879&sid=748c5ba2d0c311f7eed2870c0c47887c#p135879) by an ESP32 forum user, but I'm an ultra noob and I could not wrap my head around switching the PWM motor control code to a stepper motor one.</div>

tuliosilvajunior, 2025-01-07:
Hey, 1st of all thanks for your hard work.
I have the same situation here. 
Do you have an idea when the Winderoo will be compatible with [28BYJ-48 unipolar stepper motor with the ULN2003 motor driver?
If you need any help testing software I would like to help.

KR,

Tulio

SKendrish, 2026-01-18:
I was able to get the stepper motor to work using AccelStepper and the help of AI. The key is to work around the blocking delays in the code since the stepper run function needs to be updated frequently. Also, this was a gotcha for me, but make sure the externalbutton pin does not match your input pins for your stepper driver.