mwood77/winderoo #70: Wifi captive page issue

SpookleZz, 2024-10-18:
Hi, Love this project so far, however, I dont get any captive pages appear when connecting to the winderoo setup wifi net work. Using android version 14.

SpookleZz, 2024-10-18:
Quick update, captive portal launches on an Iphone I've dug out, so must be an android 14 issue

mwood77, 2024-10-19:
Hmm, this is an issue I've seen with android devices before. It's a long running issue with Android's (lack of) mDNS support (resolving `.local` domains on your network).

There's a very large issue tracker here talking about it: https://issuetracker.google.com/issues/140786115

It's marked as fixed, but I think that only applies to Android 14. From there, it's dependent on your devices OEM pushing that update. Who made the phone that worked, and the phone that didn't?

closed 2024-11-29 by mwood77