mwood77/winderoo #78: Console showing 193391][E][vfs_api.cpp:105] open(): /littlefs/index.html does not exist, no permits for creation

tuliosilvajunior, 2025-02-18:
It's happening some random reboots, and in the console every page refresh the message below appears.

193391][E][vfs_api.cpp:105] open(): /littlefs/index.html does not exist, no permits for creation
[STATUS] - Motor turning clockwise
[STATUS] - Motor turning clockwise
[195080][E][vfs_api.cpp:105] open(): /littlefs/assets/i18n/en-US.json.gz does not exist, no permits for creation
[195190][E][vfs_api.cpp:105] open(): /littlefs/assets/i18n/pt-BR.json.gz does not exist, no permits for creation
[ERROR] - Failed to get time from Worldtime API

mwood77, 2025-02-19:
These messages are mostly OK - I'll breakdown what's happening though:

```
193391][E][vfs_api.cpp:105] open(): /littlefs/index.html does not exist, no permits for creation
     - this message is fine and can be ignored
```

```
[195080][E][vfs_api.cpp:105] open(): /littlefs/assets/i18n/en-US.json.gz does not exist, no permits for creation
[195190][E][vfs_api.cpp:105] open(): /littlefs/assets/i18n/pt-BR.json.gz does not exist, no permits for creation
     - this message states that it cannot find certain translations
```

```
[ERROR] - Failed to get time from Worldtime API
     - this is likely the problem. When this message shows up, it means the remote NTP server, "worldtime" has rate limited your winder; it isn't returning the time.
```

I'm in the process of adding some more customizations that allows you to manually set the winder's internal clock. You can see that here: https://github.com/mwood77/winderoo/pull/71

I hope to get this merged soon, and that should solve your timing issue...I hope.

tuliosilvajunior, 2025-02-21:
Is is possible to set a locally ntp server? Ex my home ntp server?
Any other clue for the LED? I'm trying to setup mine but is not working. When I plug the ground wire from the ESP32 to the OLED the ESP32 just freeze and reboot and stay freezed until I disconet the ground wire. If I use and external ground wire from an external power supply, it's not working too.

mwood77, 2025-02-24:
> Is is possible to set a locally ntp server? Ex my home ntp server?

That's an interesting idea actually 🤔 , unfortunately at this moment there isn't.

> Any other clue for the LED? I'm trying to setup mine but is not working. When I plug the ground wire from the ESP32 to the OLED the ESP32 just freeze and reboot and stay freezed until I disconet the ground wire. If I use and external ground wire from an external power supply, it's not working too.

Hmm, are you sure all of the pinouts are set correctly in the code?

Did you follow this wiring diagram?
- https://github.com/mwood77/winderoo/blob/main/docs/wiring-diagram.md#winderoo-with-oled-screen

Do you know if these parameters are correct for your OLED screen?
- https://github.com/mwood77/winderoo/blob/442278776d41fac8bb4505ff29be8cbfe71877bb/src/platformio/osww-server/src/main.cpp#L45-L49
- https://github.com/mwood77/winderoo/blob/442278776d41fac8bb4505ff29be8cbfe71877bb/src/platformio/osww-server/src/main.cpp#L106
- https://github.com/mwood77/winderoo/blob/442278776d41fac8bb4505ff29be8cbfe71877bb/src/platformio/osww-server/src/main.cpp#L1175

Specifically in the last point `0x3C` is an address. Are you sure that's compatible with yours too?

closed 2025-02-24 by mwood77

tuliosilvajunior, 2025-03-13:
Hey bro I just change some stuffs to use my personal ntp server or you can use the ntp.pool.org that is more reliable then the [world time.](http://worldtimeapi.org/api/ip)
On platform.io

lib_deps = 
	esphome/AsyncTCP-esphome@^1.2.2
	esphome/ESPAsyncWebServer-esphome@^2.1.0
	Wire
	https://github.com/tzapu/WiFiManager.git
	https://github.com/bblanchon/ArduinoJson.git
	fbiego/ESP32Time@^2.0.0
	adafruit/Adafruit SSD1306@^2.5.9
	electromagus/ESPMX1508@^1.0.5
	dawidchyrzynski/home-assistant-integration@^2.1.0
        arduino-libraries/NTPClient@^3.2.1 //added this lib

on the main.cpp
I added 
#include <NTPClient.h>
#include <WiFiUdp.h>

remove the:
String timeURL = "http://worldtimeapi.org/api/ip";

and change the to add the proper timezone

void setup() {
    WiFi.mode(WIFI_STA);
    Serial.begin(115200);
    setCpuFrequencyMhz(160);

    //(I live in Brazil so GMT -3)
    timeClient.setTimeOffset(-10800);  // GMT-3 offset in seconds (-3 * 60 * 60)

The GetTime fuction was changed and the last two lines (http.end(); and the extra closing brace) should be removed since we're using NTP now instead of HTTP.

void getTime() {
    timeClient.begin();
    timeClient.update();
    
    time_t epochTime = timeClient.getEpochTime();
    struct tm *ptm = gmtime ((time_t *)&epochTime);
    
    int currentYear = ptm->tm_year + 1900;
    int currentMonth = ptm->tm_mon + 1;
    int currentDay = ptm->tm_mday;
    int currentHour = timeClient.getHours();
    int currentMinute = timeClient.getMinutes();
    int currentSecond = timeClient.getSeconds();
    
    Serial.printf("[TIME] - Date: %d-%02d-%02d Time: %02d:%02d:%02d\n", 
        currentYear, currentMonth, currentDay,
        currentHour, currentMinute, currentSecond);
    
    rtc.setTime(currentSecond, currentMinute, currentHour, currentDay, currentMonth, currentYear);
    
    timeClient.end();
}

mwood77, 2025-03-19:
@tuliosilvajunior #82 will introduce NTP based timekeeping. The current branch has a functional implementation if you care to try it out. Home Assistant integration will come soon.

mwood77, 2025-03-21:
@tuliosilvajunior NTP timekeeping has been merged into `main`.