mwood77/winderoo #79: New install - getting a 404 resource not found when browsing to Winderoo on local network

loiphin, 2025-02-20:
Hi,

I did an install using the provided instructions to a ESP32 with VS Code and PlatformIO. That all appears to have gone well.

I can then browse to the Winderoo via its local Access Point. I then setup my local wireless network at home, and then reset the ESP32.

It comes up on my wifi network as I can ping it, but when trying to connect to http://winderoo.local, I get the following web page:

Winderoo

404 - Resource Not found

Any ideas what I can try to help debug this ?

loiphin, 2025-02-20:
Some more info from your monitor setup:

Hard resetting via RTS pin...
--- Terminal on COM8 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
*wm:AutoConnect 
*wm:Connecting to SAVED AP: fish***
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 172.16.3.175
[STATUS] - LittleFS mounted
[STATUS] - connected to saved network
[   807][E][vfs_api.cpp:105] open(): /littlefs/settings.json does not exist, no permits for creation
[STATUS] - Failed to open configuration file, returning empty result
[STATUS] - mDNS started
[ERROR] - Failed to get time from Worldtime API
[ 36573][E][vfs_api.cpp:105] open(): /littlefs/index.html does not exist, no permits for creation <---- I BROWSE TO THE WINDERO HERE
[ 36575][E][vfs_api.cpp:105] open(): /littlefs/index.html.gz does not exist, no permits for creation
[ 36764][E][vfs_api.cpp:105] open(): /littlefs/favicon.ico does not exist, no permits for creation
[ 36766][E][vfs_api.cpp:105] open(): /littlefs/favicon.ico.gz does not exist, no permits for creation
[ 36772][E][vfs_api.cpp:105] open(): /littlefs/favicon.ico/index.html does not exist, no permits for creation
[ 36781][E][vfs_api.cpp:105] open(): /littlefs/favicon.ico/index.html.gz does not exist, no permits for creation

loiphin, 2025-02-20:
Some more testing using another board ...

Hard resetting via RTS pin...
--- Terminal on COM11 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
e*wm:AutoConnect 
*wm:No wifi saved, skipping
*wm:AutoConnect: FAILED for  13 ms
*wm:StartAP with SSID:  Winderoo Setup
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal
[STATUS] - WiFi Config Portal running
[ 39654][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 40265][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 40348][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 43903][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 64691][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 68090][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 74418][E][WebServer.cpp:649] _handleRequest(): request handler not found
*wm:3 networks found
[ 84013][E][WebServer.cpp:649] _handleRequest(): request handler not found
*wm:Connecting to NEW AP: fishnet
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:Connect to new AP [SUCCESS] 
*wm:Got IP Address:
*wm:172.16.3.185
[STATUS] - slow blink
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
*wm:AutoConnect 
*wm:Connecting to SAVED AP: fis***
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 172.16.3.185
./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
E (895) esp_littlefs: mount failed,  (-84)
E (897) esp_littlefs: Failed to initialize LittleFS
[STATUS] - LittleFS mounted
[STATUS] - connected to saved network
[  2336][E][vfs_api.cpp:105] open(): /littlefs/settings.json does not exist, no permits for creation
[STATUS] - Failed to open configuration file, returning empty result
[STATUS] - mDNS started
[ERROR] - Failed to get time from Worldtime API
[  9955][E][vfs_api.cpp:105] open(): /littlefs/index.html does not exist, no permits for creation
[  9957][E][vfs_api.cpp:105] open(): /littlefs/index.html.gz does not exist, no permits for creation

mwood77, 2025-02-20:
Looking at the logs, like here:

```
[ 9955][E][vfs_api.cpp:105] open(): /littlefs/index.html does not exist, no permits for creation
[ 9957][E][vfs_api.cpp:105] open(): /littlefs/index.html.gz does not exist, no permits for creation

and here

[ 36573][E][vfs_api.cpp:105] open(): /littlefs/index.html does not exist, no permits for creation <---- I BROWSE TO THE WINDERO HERE
[ 36575][E][vfs_api.cpp:105] open(): /littlefs/index.html.gz does not exist, no permits for creation
[ 36764][E][vfs_api.cpp:105] open(): /littlefs/favicon.ico does not exist, no permits for creation
[ 36766][E][vfs_api.cpp:105] open(): /littlefs/favicon.ico.gz does not exist, no permits for creation
[ 36772][E][vfs_api.cpp:105] open(): /littlefs/favicon.ico/index.html does not exist, no permits for creation
[ 36781][E][vfs_api.cpp:105] open(): /littlefs/favicon.ico/index.html.gz does not exist, no permits for creation
```

I'd wager a guess that you didn't flash the filesystem. You have to do that as well as "upload the code." Specifically step 7 here:
https://github.com/mwood77/winderoo/blob/main/docs/install-software.md#flashing-your-microcontroller

loiphin, 2025-02-20:
OMG.... thats embarrassing! Sorry to waste your time.

In your docs though, would you state what point 1. and point 2. mean in the diagram, so its clear for everyone?

Thanks again,

loiphin.

mwood77, 2025-02-20:
Sure, I can clarify that.

Do you mean, it would be better to illustrate the difference between steps 6 and 7 - which look very similar?

loiphin, 2025-02-20:
Yes please :) I think my brain thought okay, i have done everything , uploaded the firmware, and it should work. And stop reading properly :/....  I guess it wasnt clear there are two partitions, code and data.

loiphin, 2025-02-20:
Also , I am getting about 9RPM as speed, with no PWM. Do you work out the "turns per day" based on time, and an estimate of RPM ? Whats the RPM you worked out.

mwood77, 2025-02-20:
> Yes please :) I think my brain thought okay, i have done everything , uploaded the firmware, and it should work (and then I stop reading properly :/) .... I guess it wasnt clear there are two partitions, code and data.

👍  I'll squeeze this into an existing PR.

> Also , I am getting about 9RPM as speed, with no PWM. Do you work out the "turns per day" based on time, and an estimate of RPM ? Whats the RPM you worked out.
> 
> So I am getting 26 minutes estimate, but in reality it should be 11minutes for 100 turns per day (9RPM), if I understand right.
> 
> Any chance we could configure RPM and PWN in an advanced settings tab ? :)
> 
> Wolf, a brand of winders rotates at 6RPM. Dont know if there is any scientific merit in that?

If you're willing to beta test, you can pull the code off this branch:
- https://github.com/mwood77/winderoo/pull/71

That branch contains configurable:
- clock (the current NTP server 429's Winderoo pretty badly)
- winding per-cycle time
- winding per-cycle pause time
- motor rotation timing

loiphin, 2025-02-20:
Awesome, just tried it now and it seems to work well :) Only difference I have found is that clockwise /counter clockwise is reversed in this build. No biggy, I can reverse the wiring to accomodate.

Thank you ever so much for building this project :) I have loved every minute of building it. It scratches so many itches of mine, mechanics, electronics, horologyl, 3d printing, and coding :)

mwood77, 2025-02-20:
Great, glad you've enjoyed it!

Please @me if you find any bugs in the code from that branch. Part of it was a significant rewrite, so I'm somewhat apprehensive to merge it.

closed 2025-03-19 by mwood77