mwood77/winderoo #80: ESP32 Rebooting when disabling the OLED Screen - SSD1306 OLED

tuliosilvajunior, 2025-02-22:
I'm facing an issue when I connect the SSD1306 OLED nothing appears and if I try to disable using the GUI the ESP32 simply reboots.
Sometimes it froozen and it just but if I remove the OLED.
I tested with 2 different OLED and they are working. 

E (149552) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (149552) task_wdt:  - async_tcp (CPU 0/1)
E (149552) task_wdt: Tasks currently running:
E (149552) task_wdt: CPU 0: IDLE
E (149552) task_wdt: CPU 1: IDLE
E (149552) task_wdt: Aborting.

abort() was called at PC 0x4010083c on core 0

Backtrace: 0x40083ccd:0x3ffbeccc |<-CORRUPTED

ELF file SHA256: 4d44ce74793736e9

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
*wm:AutoConnect 
*wm:Connecting to SAVED AP: myiot
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 192.168.1.197
[STATUS] - LittleFS mounted
[STATUS] - connected to saved network
[STATUS] - mDNS started
[STATUS] - HA Configured - Will attempt to connect to MQTT broker

mwood77, 2025-02-24:
I realized I posted my reply in the other issue, so here's a cross-post:

<hr>

> Any other clue for the LED? I'm trying to setup mine but is not working. When I plug the ground wire from the ESP32 to the OLED the ESP32 just freeze and reboot and stay freezed until I disconet the ground wire. If I use and external ground wire from an external power supply, it's not working too.

Hmm, are you sure all of the pinouts are set correctly in the code?

Did you follow this wiring diagram?
- https://github.com/mwood77/winderoo/blob/main/docs/wiring-diagram.md#winderoo-with-oled-screen

Do you know if these parameters are correct for your OLED screen?
- https://github.com/mwood77/winderoo/blob/442278776d41fac8bb4505ff29be8cbfe71877bb/src/platformio/osww-server/src/main.cpp#L45-L49
- https://github.com/mwood77/winderoo/blob/442278776d41fac8bb4505ff29be8cbfe71877bb/src/platformio/osww-server/src/main.cpp#L106
- https://github.com/mwood77/winderoo/blob/442278776d41fac8bb4505ff29be8cbfe71877bb/src/platformio/osww-server/src/main.cpp#L1175

Specifically in the last point `0x3C` is an address. Are you sure that's compatible with yours too?

tuliosilvajunior, 2025-02-24:
> I realized I posted my reply in the other issue, so here's a cross-post:
> 
> > Any other clue for the LED? I'm trying to setup mine but is not working. When I plug the ground wire from the ESP32 to the OLED the ESP32 just freeze and reboot and stay freezed until I disconet the ground wire. If I use and external ground wire from an external power supply, it's not working too.
> 
> Hmm, are you sure all of the pinouts are set correctly in the code?
> 
> Did you follow this wiring diagram?
> 
> * https://github.com/mwood77/winderoo/blob/main/docs/wiring-diagram.md#winderoo-with-oled-screen
> 
> Do you know if these parameters are correct for your OLED screen?
> 
> * [winderoo/src/platformio/osww-server/src/main.cpp](https://github.com/mwood77/winderoo/blob/442278776d41fac8bb4505ff29be8cbfe71877bb/src/platformio/osww-server/src/main.cpp#L45-L49)
>       
>       
>           Lines 45 to 49
>         in
>         [4422787](/mwood77/winderoo/commit/442278776d41fac8bb4505ff29be8cbfe71877bb)
>       
>     
>     
>       
>   
>           
>             
>              bool OLED_INVERT_SCREEN = false; 
>           
>   
>           
>             
>              bool OLED_ROTATE_SCREEN_180 = false; 
>           
>   
>           
>             
>              int SCREEN_WIDTH = 128; // OLED display width, in pixels 
>           
>   
>           
>             
>              int SCREEN_HEIGHT = 64; // OLED display height, in pixels 
>           
>   
>           
>             
>              int OLED_RESET = -1; // Reset pin number (or -1 if sharing Arduino reset pin)
> * [winderoo/src/platformio/osww-server/src/main.cpp](https://github.com/mwood77/winderoo/blob/442278776d41fac8bb4505ff29be8cbfe71877bb/src/platformio/osww-server/src/main.cpp#L106)
>       
>       
>            Line 106
>         in
>         [4422787](/mwood77/winderoo/commit/442278776d41fac8bb4505ff29be8cbfe71877bb)
>       
>     
>     
>       
>   
>           
>             
>              Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);
> * [winderoo/src/platformio/osww-server/src/main.cpp](https://github.com/mwood77/winderoo/blob/442278776d41fac8bb4505ff29be8cbfe71877bb/src/platformio/osww-server/src/main.cpp#L1175)
>       
>       
>            Line 1175
>         in
>         [4422787](/mwood77/winderoo/commit/442278776d41fac8bb4505ff29be8cbfe71877bb)
>       
>     
>     
>       
>   
>           
>             
>              if (!display.begin(SSD1306_SWITCHCAPVCC, 0x3C))
> 
> Specifically in the last point `0x3C` is an address. Are you sure that's compatible with yours too?

Hey I checked everything replaced all cables and the power supply. Everything now is working like a charm. Thank you again about your hard work.
Any plan to incorporate multiples winderoo units to be managed together?

closed 2025-02-24 by tuliosilvajunior

mwood77, 2025-02-24:
I hit a snag getting multiple units networked together. I plan to look at it again when I have time to do a significant refactor . SO it's in the works, sometime soon ™  😉